fbiego/ESP32Time #1: GetMilliseconds()

JRVeale, 2021-02-11:
Nice wrapper library!

I see there's no GetMilliseconds() function. Seeing as the user can input ms when setting the current time, it would be nice to be able to get the ms out as well.

Is this possible? Thanks

fbiego, 2021-02-14:
Updated
commit [8c6d08d](https://github.com/fbiego/ESP32Time/commit/8c6d08d78cfbdfa12001710deb3f71400d531771)

closed 2021-02-14 by fbiego

JRVeale, 2021-02-15:
```
long ESP32Time::getMillis(){
	struct timeval tv;
	gettimeofday(&tv, NULL);
	return tv.tv_sec;
}
```

Shouldn't the return value be `tv.tv_usec`?

fbiego, 2021-02-15:
the function above returns the epoch seconds
I've made the following changes;
- getEpoch() `return tv.tv_sec`
- getMillis() `return tv.tv_usec/1000`
- getMicros() `return tv.tv_usec`