fbiego/ESP32Time #6: Lost Time

ebelouet, 2021-04-01:
HI,

You example work very well but I have 6 minutes more in one day, where is mistake ?

Sincerely
Eric

fbiego, 2021-04-01:
The internal RTC is not accurate
> Internal RTC clock frequency error is about 5%

It is recommended you set/sync the time periodically

ebelouet, 2021-04-01:
thank, best way  is to put external crysral, but your example takie it to run.

Sincerely
Eric

ebelouet, 2021-04-05:
Hi,
Add external crystal option will be good for your library ....to have a  vervy good time solution

fbiego, 2021-04-05:
the clock source cannot be changed.
[arduino-esp32/issues/3964](https://github.com/espressif/arduino-esp32/issues/3964)
to change it the `arduino-esp32` library has to be recompiled

closed 2023-08-01 by fbiego