fbiego/ESP32Time #16: 'tm timeinfo' has initializer but incomplete type

Albalate, 2022-02-11:
hello.
When trying to compile the example I get the following error:

C:\Users\Usuario\Documents\Arduino\esp32\prueba1\prueba1.ino: In function 'void loop()':
prueba1:64:13: error: variable 'tm timeinfo' has initializer but incomplete type
   struct tm timeinfo = rtc.getTimeStruct();
             ^~~~~~~~
prueba1:64:42: error: invalid use of incomplete type 'struct tm'
   struct tm timeinfo = rtc.getTimeStruct();
                                          ^
In file included from C:\Users\Usuario\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\cores\esp32/Stream.h:26,
                 from C:\Users\Usuario\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\cores\esp32/Arduino.h:160,
                 from sketch\prueba1.ino.cpp:1:
C:\Users\Usuario\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\cores\esp32/Print.h:94:25: note: forward declaration of 'struct tm'
     size_t print(struct tm * timeinfo, const char * format = NULL);
                         ^~
exit status 1
variable 'tm timeinfo' has initializer but incomplete type

IDE preferences are:
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
https://arduino.esp8266.com/stable/package_esp8266com_index.json
https://dl.espressif.com/dl/package_esp32_index.json

fbiego, 2022-02-12:
Refer to issue #11 or #15

closed 2022-02-12 by fbiego

Albalate, 2022-02-12:
Thanks. Is good solutions!