fbiego/ESP32Time #21: Typo in /examples/esp32_time.ino causes strange behaviour

pietropizz, 2022-09-26:
I tried your great library and used esp32_time.ino for that. In the following line is a typo:
rtc.setTime(30, 24, 15, 17, 1, 2042);  // 17th Jan 2021 15:24:30
With manually rtc.setTime() and the year greater than 2035 you get a strange behaviour when you use the ntp function afterwards:
configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);
In this case the year sets up to 2088, the rest of date and time is updated correctly.
If you use a year 2035 or lower, the NTP corrects to the actual year as well.
I am not a professional C++ developer, therefore I can not analyse your code in detail.

Calcousin55, 2024-03-17:
Might be related [https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/system_time.html#year-2036-and-2038-overflow-issues](url)

fbiego, 2024-03-26:
The overflow is handled internally by the library using a flag, so you need to re-set the time using the library function after configuring with ntp
https://github.com/fbiego/ESP32Time/blob/56c7cca5838249db5bff3cbc4f67e7da0f91b126/examples/esp32_time/esp32_time.ino#L36-L41

closed 2024-03-26 by fbiego