fbiego/ESP32Time #27: Missing initializer for member "overflow"

Nanowires, 2023-07-03:
Hi,

I'm using the IDE sloeber and get the compiler warning:
`Member 'overflow' was not initialized in this constructor`

Also when I'm using the library, the overflow is always true (remark: uninitialized variables are random, bool is using an uint8_t I guess, so 254 of 255 cases).

Fix would be to explicitly initialize the value of overflow in the constructor.

In your defence, it semms like the Arduino IDE compiler will automatically initialize primitive types, but it's not specified behaviour and should be avoided ;-)

Thanks

fbiego, 2023-07-03:
Can you create a pr with the changes?

Nanowires, 2023-07-04:
Please see  #28

closed 2023-07-05 by fbiego

Nanowires, 2023-07-05:
This was fast and easy 😲 👍