fbiego/ESP32Time #29: unsigned long offset?

nicolasff, 2023-07-11:
Hello,

There seems to be a conflict between the type of the constructor parameter to `ESP32Time` and the example using a negative offset – or really anyone trying to use a time zone west of the Greenwich meridian.

The constructor needs an `unsigned long offset`: https://github.com/fbiego/ESP32Time/blob/7f84a2b9863cbe01bed3a1aecd84b283923de1e7/ESP32Time.h#L34

but one of the examples in `esp32_time_multiple.ino` uses `-3600` (a **signed** value) as the offset for GMT-1: https://github.com/fbiego/ESP32Time/blob/7f84a2b9863cbe01bed3a1aecd84b283923de1e7/examples/esp32_time_multiple/esp32_time_multiple.ino#L28

How is this supposed to work? I came to this GitHub project looking for examples because I was confused by the `unsigned long` when my time zone is UTC minus some hours, but seeing this negative value being passed to an unsigned variable did not exactly clear things up.

Thanks!

fbiego, 2023-07-11:
Clearly this is a bug, will be fixed soon

closed 2023-07-11 by fbiego

nicolasff, 2023-07-11:
Thank you! That was so quick.

commit 548dfd38 (2023-07-11): fix #29 bug