fbiego/ESP32Time #31: Hour goes forward one hour

newsweet-ui, 2023-07-16:
Hi!

I set 22 o'Clock and get 23 back.
hr contains the value 22.
You can see the output on the screenshot from the serial monitor.
What is wrong here? Do you have an idea?

Nicole

`  int sc = fieldContent.substring(20,22).toInt();
  int mn = fieldContent.substring(17,19).toInt();
  int hr = fieldContent.substring(14,16).toInt();
  int dy = fieldContent.substring(3,5).toInt();
  int mt = fieldContent.substring(6,8).toInt();
  int yr = fieldContent.substring(9,13).toInt();
  int ms =0;

  Serial.print("hr: ");
  Serial.println(hr);
                             
  rtc.setTime(sc,mn,hr,dy,mt,yr);        
              
  Serial.println(rtc.getTime("%d/%m/%Y %H:%M:%S"));`

[image]

fbiego, 2023-07-16:
The time is set in UTC
Maybe you had modified the offset to GMT+1
Check it
`Serial.println(rtc.offset);`

newsweet-ui, 2023-07-17:
> The time is set in UTC Maybe you had modified the offset to GMT+1 Check it `Serial.println(rtc.offset);`

Thank you, you are right. I had the wrong offset.

closed 2023-07-17 by newsweet-ui