fbiego/ESP32Time #42: Error message upon verify related to setTimeStruct in ESP32Time.h

Tsumomo, 2024-05-05:
I attempted to integrate this library into my latest Adafruit Matix Portal M4 project and I am getting an odd error. I went back to basics and attempted to run the same code in the esp32_time.ino example and I get the same error.

In file included from D:\Arudino\esp32_time\esp32_time.ino:3:
c:\Users\marti\Documents\Arduino\libraries\ESP32Time/ESP32Time.h:37:22: error: 'tm' has not been declared
   37 |   void setTimeStruct(tm t);
      |                      ^~
c:\Users\marti\Documents\Arduino\libraries\ESP32Time/ESP32Time.h:38:3: error: 'tm' does not name a type; did you mean 'Pm'?
   38 |   tm getTimeStruct();
      |   ^~
      |   Pm

exit status 1

Compilation error: exit status 1

After a bit of searching I can't find any other solutions, hopefully someone can assist please.

fbiego, 2024-05-06:
This is a wrapper library for ESP32 time functions, it relies on time libraries used by ESP32 core
https://github.com/fbiego/ESP32Time/blob/56c7cca5838249db5bff3cbc4f67e7da0f91b126/ESP32Time.cpp#L26-L27

Compatibility with other Arduino cores is untested

Tsumomo, 2024-05-07:
Thank you for getting back to me quickly. I'll head over to Adafruit and start with Circuit Python which seems to be what they prefer.