fbiego/ESP32Time #44: ESP32 S2 - Not Working

TCB13, 2024-08-03:
On my ESP32-S2 setting the date/time like this:

```
rtc.setTime(0, 10, 13, 3, 8, 2024);
```

Will result int: 
```
Serial.println("call day: " + String(this->nativeTime.getDay()));
Serial.println("call month: " + String(this->nativeTime.getMonth()));
Serial.println("call year: " + String(this->nativeTime.getYear()));
//...
call day: 1
call month: 0
call year: 1970
```

Any ideia why that might be?

Thanks.

TCB13, 2024-08-03:
However, following the following works and seems to be what the library does (minus the overflow stuff):

```
    struct tm t = {0, 0, 0, 0, 0, 0, 0, 0, 0};      // Initialize to all 0's
    t.tm_year = 2024 - 1900;    // This is year-1900, so 121 = 2021
    t.tm_mon = 8 - 1;
    t.tm_mday = 3;
    t.tm_hour = 14;
    t.tm_min = 11;
    t.tm_sec = 0;
    time_t timeSinceEpoch = mktime(&t);

    struct timeval tv;
    tv.tv_sec = timeSinceEpoch; 
    tv.tv_usec = 0;
    settimeofday(&tv, NULL);
```

Not sure about what's going on.

fbiego, 2024-08-03:
That's strange. I don't have ESP32-S2 to debug with, but can you provide the full code pertaining the issue?

TCB13, 2024-08-03:
> That's strange. I don't have ESP32-S2 to debug with, but can you provide the full code pertaining the issue?

Thanks for the reply.

I replaced the board with another S2 and it works, I'm sorry for bothering you.

closed 2024-08-03 by TCB13