fbiego/ESP32Time #46: ABOUT   getMillis()

weimin1668, 2025-01-08:
getMillis()   
When I use the ESP32-S3 chip development board, the milliseconds of the output is wrong with the actual one. Is it not a millisecond relationship? What is the reason ah?

Serial.println("rtc.getMillis()  getMicros");
Serial.println(rtc.getMillis());

fbiego, 2025-01-08:
Can you share how you are getting the actual one?

weimin1668, 2025-01-08:
program :
#include <ESP32Time.h>
   // 创建RTC对象实例
    //ESP32Time rtc;
ESP32Time rtc(0); 

void setup() {
  // put your setup code here, to run once:
    Serial.begin(115200);
    delay(1000);
    
    rtc.setTime(30, 52, 16, 8, 1, 2025);
}

void loop() {
  // put your main code here, to run repeatedly:
Serial.println(rtc.getEpoch());  //  It is time is UTC+8:00 . 10进制 ------rtc.getEpoch()==
Serial.println(rtc.getTime("%A, %B %d %Y %H:%M:%S")); 
Serial.print("rtc.getMillis() =====");
Serial.println(rtc.getMillis());//
Serial.print("rtc.getMicros========");
Serial.println(rtc.getMicros());//
Serial.print("millis()===========");
Serial.println(millis());
delay(2000);
}
[image]

weimin1668, 2025-01-11:
OH,OK,  Thank you very much!!

closed 2025-04-10 by fbiego